Ameba8195/Arduino #2: ERR: Cannot access mbed driver!

shmrymbd, 2016-11-02:
Hi i am using macos. having problem to upload

brainstorm, 2016-11-02:
The CH340 USB-to-serial chip access support is not implemented yet since, apparently, the upload is too slow, see issue #1.

If you feel adventurous, you could try plugging in an SWD-capable gear such as the J-Link Segger and have OpenOCD program it, as I did myself in here:

https://bitbucket.org/rebane/rtl8710_openocd/pull-requests/1/add-support-for-osx-with-preinstalled/diff

Since it's also under development (the firmware crashes for me since I still have to tweak the memory layout/flashing), I cannot help you much there either. I hope @Ameba8195 implements the UART support anyway, even if it's too slow, to give easy access to the rtl8710 to the general public.

Btw, how slow is to upload i.e 1MB of `ram_all.bin` into the RTL8710 via CH340, @Ameba8195? Did you try it out at all?

shmrymbd, 2016-11-02:
hi i'm just stuck with arduino ide .. the error ERR: Cannot access mbed driver!

brainstorm, 2016-11-02:
@Ameba8195: For instance, the ESP8266 can be programmed really fast only via UART:

https://blog.cesanta.com/esp8266-superfast-flashing-flash-esp-in-5-seconds

@shmrymbd, yes, you should wait until all this is supported and update then, sorry, come back later on ;)

andrews05, 2017-03-16:
@Ameba8195, any update on this?

maticzav, 2017-05-30:
Any progress with this?

treii28, 2017-06-05:
any progress on this? (using ubuntu linux)

astro, 2017-09-23:
Any progress?

kissste, 2018-09-19:
+1 got the same error

robertoostenveld, 2018-09-19:
+1 got the same error

ricardoper, 2019-01-02:
Ubuntu 18.04 - Arduino 1.8.8 - Same error...

Any alternative to make it working from Ubuntu or any Linux OS?

kissste, 2019-01-02:
> Ubuntu 18.04 - Arduino 1.8.8 - Same error...
> 
> Any alternative to make it working from Ubuntu or any Linux OS?

This is the only solution that I am aware of:
https://github.com/Ameba8195/Arduino/issues/2#issuecomment-257833909

temach, 2019-03-15:
Not sure how relevant this is, but anyway.

The mbed/ameba approach means the microcontroller appears as a flash drive in the os. The arduino after compiling and transforming the binary copies the binary to this flash drive. The commands it uses can be found at the bottom of platform.txt which on my system is at ~/.arduino15/packages/RAK/hardware/ameba/2.0.2/platform.txt

Looking into that file I see this line:
tools.ameba_tools.upload.pattern.linux="{runtime.tools.ameba_tools.path}/upload_dap_linux" "{runtime.tools.ameba_tools.path}/ram_all.bin"{upload.verbose}

So it uses the program "upload_dap_linux", this program is located in:
~/.arduino15/packages/RAK/tools/ameba_tools/1.0.9/upload_dap_linux 
But the most important part is that its source is located nearby:
~/.arduino15/packages/RAK/tools/ameba_tools/1.0.9/tools/linux/src/upload_dap_linux.cpp 
(also availiable here: https://github.com/RAKWireless/CREATOR-Arduino-SDK/blob/master/ameba_tools_linux/tools/linux/src/upload_dap_linux.cpp )

This source contains the actual error message: "Cannot access mbed driver!"
Looks like the program expects the mbed/ameba usb drive to be mounted at an unusual location, and if it does not find it - it quits with error.

miterfrants, 2022-08-09:
rename your `/Volumes/DAPLINK` to `/Volumes/MBED` @shmrymbd