Ameba8195/Arduino #6: I2C endTransmission crashing board

gtalusan, 2017-03-01:
Hi,

I have a PADI IoT Stamp that I've soldered onto a breakout.  I have it connected to an Si1445 UV sensor via I2C with SCL=GC5 and SDA=GC4.  My sketch is the example si1145test from Adafruit:

```
#include <Wire.h>
#include "Adafruit_SI1145.h"

Adafruit_SI1145 uv = Adafruit_SI1145();

void setup() {
  Serial.begin(9600);
  
  Serial.println("Adafruit SI1145 test");
  
  if (! uv.begin()) {
    Serial.println("Didn't find Si1145");
    while (1);
  }

  Serial.println("OK!");
}

void loop() {
  Serial.println("===================");
  Serial.print("Vis: "); Serial.println(uv.readVisible());
  Serial.print("IR: "); Serial.println(uv.readIR());
  
  // Uncomment if you have an IR LED attached to LED pin!
  //Serial.print("Prox: "); Serial.println(uv.readProx());

  float UVindex = uv.readUV();
  // the index is multiplied by 100 so to get the
  // integer index, divide by 100!
  UVindex /= 100.0;  
  Serial.print("UV: ");  Serial.println(UVindex);

  delay(1000);
}
```
The sensor is verified to work when connected to an Arduino Genuino and Pro Mini clone.

After some debugging, I've found that the PADI Stamp board is crashing in the first call to `Wire::endTransmission` from `uv.begin()` in `setup`.

It looks like that `i2c_write` is doing something wrong, but I can't tell what..

Ameba8195, 2017-03-01:
Thanks for report this issue. We'll look into this issue and feedback later

gtalusan, 2017-03-01:
Thanks, I resoldered my connections for GC4 and GC5 and it started working.  I have an I2C OLED working too.

[image]

It would be nice if all of the pins were broken out in rtl8710/variant.cpp.

closed 2017-03-01 by gtalusan