Ameba8195/Arduino #8: SDK 2.0.1 cannot install tools 1.0.9

pilnikov, 2017-03-18:
In arduino 1.8.1 on win10x64 install new version of ameba (2.0.1) ended with next messages
Index error: could not find referenced tool name=ameba_tools version=1.0.9 packager=realtek
The archive with "tools" in _\staging\ packages\_  not present. In _\packages\realtek_ _\hardware_\ present only, _\tools\_ is not present

pilnikov, 2017-03-19:
I had to manually create the folder and put the contents of arhive file "ameba_tools_windows-1.0.9.tar.gz" there

Ameba8195, 2017-03-20:
We've noticed this issue and will fix it ASAP. Thanks for feedback this issue.

Ameba8195, 2017-03-20:
Hi, we have solved this issue. Please try remove ameba in board manager and install again.

pilnikov, 2017-03-20:
everything worked! Thanks

closed 2017-03-20 by pilnikov