Ameba8195/Arduino #19: RTL8710 arduino

ITstreet1, 2022-02-26:
I have successfully uploaded a sketch to an RTL8720D. The goal is to use LG RX/TX with a serial adapter. My question is, do I use the same approach with RTL8710? RTL00 module.
If so, are the LOG pins in the same places? GE1 and GE0?

kissste, 2022-02-26:
I don't think so. The firmware, in ROM, on RTL8710 does not support it.

kissste, 2022-02-26:
I would not call it "any sketch" - it needs to be your sketch that is going to implement your a method/ implementation of receiving a code from serial port, flashing it and upon reboot running it. Defacto your own bootloader

Otherwise you are correct.

ITstreet1, 2022-02-26:
Wait a minute...
Here is a support for the RTL8710, which is not support at all. I missed something...

To upload a "bootloader" sketch I need to choose the SWD option in Arduino IDE.
-There is no option for the SWD
-I have not the slightest idea what kind of that sketch should be. Here, there is not a single line that explains it. Not to mention that "bootloader" sketch as an example.

Making support for an MCU without these parts is not support at all. Did any make this work? I haven't found not a single line someone made this work. 

When you make support for an MCU, it should have a complete package. Otherwise, it is incomplete.

kissste, 2022-02-26:
I agree, 100%, don't waste your time with the chip. Stick with ESP8266 or ESP32

ITstreet1, 2022-02-26:
Then, what are the repos at your GitHub for?

kissste, 2022-02-26:
I played with it back in ~2017 or so, had high hopes, good hardware, but no support from the vendor, and too many dead ends, resulted no adoption (in the hobby segment)

closed 2022-03-08 by ITstreet1